TypeScriptSystemVerilog/TSSV #8: output directory needs exist before run node out/test/test_FIR.js

aigcic-richard, 2024-03-31:
$ node out/test/test_FIR.js
$ cat sv-examples/test_FIR_output/myFIR.sv

if manually remove test_FIR_output, command node out/test/test_FIR.js will fail.

Error: ENOENT: no such file or directory, open 'sv-examples/test_FIR_output/myFIR3.sv'
    at writeFileSync (node:fs:2342:20)
    at file:///Users/richardzhai/github/richard/TSSV/out/test/test_FIR.js:19:5
    at ModuleJob.run (node:internal/modules/esm/module_job:222:25)
    at async ModuleLoader.import (node:internal/modules/esm/loader:323:24)
    at async loadESM (node:internal/process/esm_loader:28:7)
    at async handleMainPromise (node:internal/modules/run_main:113:12) {
  errno: -2,
  code: 'ENOENT',
  syscall: 'open',
  path: 'sv-examples/test_FIR_output/myFIR3.sv'
}

seems code inside js needs first check if output folder exists or not, if not, create folder before run generate output code.

avonancken, 2024-04-04:
Made this change to fix this issue:
`ts/test/test_FIR.ts`
```diff
import { FIR } from 'tssv/lib/modules/FIR'
- import { writeFileSync } from 'fs'
+ import { writeFileSync, mkdirSync } from 'fs'

+ try {
+   mkdirSync('sv-examples/test_FIR_output')
+ } catch (e) {}

const myFir = new FIR({ name: 'myFIR', numTaps: 4, coefficients: [1n, 2n, 3n, 4n] })
try {
```

avonancken, 2024-04-04:
Closing as fixed.

closed 2024-04-04 by avonancken